Sensirion/embedded-sps #14: checksum error

khongco44, 2019-01-14:
Hi
I am trying to read values of the sps30 via I2C with esp8266 and the arduino IDE.
i used library embedded-sps and embedded-common to programing, but when my code run at function

`    while (sps30_probe() != 0) {
        Serial.print("probe failed\n");
        delay(1000);
    }`
then serial shows this message: "probe failed"

I found the problem and i saw, inside function:

`int16_t sensirion_i2c_read_bytes(uint8_t address, uint8_t *data, int64_t num_words) {
    int16_t ret;
    int64_t i, j;
    int64_t size = num_words * (SENSIRION_WORD_SIZE + CRC8_LEN);
    int64_t word_buf[SENSIRION_MAX_BUFFER_WORDS];
    uint8_t * const buf8 = (uint8_t *)word_buf;

    ret = sensirion_i2c_read(address, buf8, size);
    if (ret != STATUS_OK)
        return ret;
    /* check the CRC for each word */
    for (i = 0, j = 0; i < size; i += SENSIRION_WORD_SIZE + CRC8_LEN) {

        ret = sensirion_common_check_crc(&buf8[i], SENSIRION_WORD_SIZE,
                                         buf8[i + SENSIRION_WORD_SIZE]);
        if (ret != STATUS_OK)
            return ret;

        data[j++] = buf8[i];
        data[j++] = buf8[i + 1];
    }
    return STATUS_OK;
}`

`uint8_t sensirion_common_generate_crc(uint8_t *data, int64_t count)
{
    int64_t current_byte;
    uint8_t crc = CRC8_INIT;
    uint8_t crc_bit;

    /* calculates 8-Bit checksum with given polynomial */
    for (current_byte = 0; current_byte < count; ++current_byte) {
        crc ^= (data[current_byte]);
        for (crc_bit = 8; crc_bit > 0; --crc_bit) {
            if (crc & 0x80)
                crc = (crc << 1) ^ CRC8_POLYNOMIAL;
            else
                crc = (crc << 1);
        }
    }
    return crc;
}`
with buf8[30],buf8[31] and buf8[32] checksum value not corect
when use buf8[30] ,buf8[31] to calculate checksum value, the result don't match with buf8[32],
program cannot break out while loop.
I tried with 3 board: arduino, esp32, esp8266 inside arduino IDE but error still same.
I don't know how can i fix
Can i remove checksum function ?
Plz help me, thanks

abrauchli, 2019-01-14:
Hi @khongco44 

Did you make sure that the sensors starts in i2c mode and not in UART mode (pin4 pulled to ground)?

Could you print the content of the buffer (I'm assuming that it fails to retrieve the serial number)? Do you have a logic analyzer (e.g. from salae) that you could debug the i2c communication with?

If you have a CRC error it's unlikely that it consistently appears at the same buffer index but are there other devices on the i2c bus that could cause crosstalk?

abrauchli, 2019-01-14:
Oh wait, are you running into Arduino's maximal buffer size issue discussed in #7? If so, try using the UART driver (https://github.com/Sensirion/embedded-uart-sps) or the software-i2c library which doesn't suffer from the limited buffer.

michapr, 2019-01-17:
@khongco44 : if you use ESP you can modify the wire.h lib.

change: #define BUFFER_LENGTH 64  (was 32)

Please remember that this will take more RAM from your device and can be overwritten after Arduino update.

khongco44, 2019-01-18:
@michapr 
thanks bro, I will try it

khongco44, 2019-01-18:
Hi @michapr 
i changed #define BUFFER_LENGTH 64, but still same, read i2c

michapr, 2019-01-19:
I have checked it with Arduino 1.8.7 - here it is working as I wrote, editing only Wirhe.h
My small program is working now, and the provided sample too.

What happen if you will initialize the SPS only and ask for the ready data flag? Is it working?
I think you have checked the device with i2c scanner ? ;)

Best regards
Michael

michapr, 2019-01-20:
Good news - will work for ESP8266 in future release
The request to increase size in library was merged (after I have requested again ;) )
https://github.com/esp8266/Arduino/pull/3576#pullrequestreview-194377996

paultanner, 2019-01-20:
Unfortunately with Arduino 1.8.8 on Leonardo (and the mod toWire.h) it's still not working.  The problem is the same as above: no response from the SPS-30 seen on Salae.  The command looks the same so, given that it works on other chips,  I'm probably looking at something like a memory overflow.

michapr, 2019-01-22:
@paultanner You can try my simple sketch - maybe it will help.
https://smarthome-work.com/blog/?particulate-matter-sensor-sps30-and-arduino

It is working with small buffer too - but then only for the first 5 values.

paultanner, 2019-01-23:
Many thanks @michapr 
This worked with 32 byte buffer but broke when I changed Wire.h to 64 byte buffer.
I looked again for twi.h and found it this time in /utility in the Wire folder (Arduino 1.6.12). 
I changed it as follows and your mini example now works.
#define TWI_BUFFER_LENGTH 64
So evidently we were dealing with an overrun issue.
At the moment the original code from this repo is still failing at sps30_probe().
It should be much easier to track this down now. 
Thanks again.

justb4, 2019-02-03:
Also struggled to get SPS30 working with a [ThingsNetwork UNO Node ](https://www.thethingsnetwork.org/docs/devices/uno/)(which is a Leonardo in fact).  

Found this repo: https://github.com/paulvha/sps30 . Well-documented (read also the [sps30.odt ](https://github.com/paulvha/sps30/blob/master/sps30.odt) doc!) . According to the author reading more than 5 values via I2C is never possible _"... as the amount of data that is sent by the SPS30 in I2C is always 32 bytes."_.

Got SPS30 with Leonardo via I2C (according to the author UART is not possible with UNO R3), with the following config:

* connection: Leonardo has SCL, SDA pins, used 10kOhm pull-up resistors
* `sps30.h` line 44,50:  disabled ` #define INCLUDE_UART 1` and `#define INCLUDE_SOFTWARE_SERIAL 1`
* `sps30.h` line 55: added `|| defined(__AVR_ATmega32U4__) || defined(__AVR_ATmega16U4__)` to enable `#define SMALLFOOTPRINT 1`

Ran the example [Example1_sps30_BasicReadings.ino](https://github.com/paulvha/sps30/blob/master/examples/Example1_sps30_BasicReadings/Example1_sps30_BasicReadings.ino) ok! See output below:

[image]

Credits and kudos to @paulvha , as he did a very nice job, making a compact driver lib with ample examples and documentation!

justb4, 2019-02-03:
@paulvha yes, much nicer output with `DEBUG 0`. Was dealing with many issues at the same time, including the wiring/pins/pull-up resistors, so had full debugging on. Eventually targeting to connect SPS30 to [PyCom LoPy4](https://pycom.io/product/lopy4/), which will be even more of a challenge as I found no drivers nor folks using that combo.

paulvha, 2019-02-03:
Thanks

Just to make clear and good news !!!  : I stand corrected !

With I2C communication you CAN get all the information. The issue is indeed the 32 bytes I2C buffer length on Arduino and ESP8266 (maybe others). If the length send is more than this buffer size it will send will send a NACK, thus stopping the SDS30 from sending more.

I have adjusted my library that it will check and whether the I2C buffer size is at least 64. It will request and display the full amount else it will stick to the MASS values only. My beta test version worked as expected.

Now need to update my documentation, examples and test. Expect the new version to be available tomorrow.

regards,
Paul

justb4, 2019-02-04:
@paulvha yes, I was not clear: the I2C limitation is indeed only for certain boards (Arduino (UNO, Leonardo), ESP8266,...). If that is really the case, and possible workarounds then that should be documented in the Sensirion SPS30 docs/specs. We can take issues specific to your driver to your repo, https://github.com/paulvha/sps30. And don't feel rushed :-), you already did an excellent job!

Hopefully Sensirion folks follow this discussion and your repo, so eventually all efforts find their way to usable drivers for our AQ-communities. That would be also be beneficial to Sensirion as AQ communities like [LuftDaten](https://luftdaten.info/) are constantly on the lookout for better/more accurate PM measurements.

paulvha, 2019-02-04:
Just posted the updated driver on the github

Based on some observations, feedback and learning I have made changes to the driver. The most important one is around I2C You can get all the SPS30 data in I2C communication, but you need to extend the I2C receive buffer in Wire.h. The driver will now check whether that buffer is at least 64 bytes and then provide all the values, else it will only provide the 4 MASS values.
A new call has been added (I2c_expect()) which will return the expected number of valid values to receive from the SPS30 (either 4 or 10).

If have updated the samples, documentation and added new example as indicated in the readme file

justb4, 2019-02-04:
@paulvha the above version works for me (Leonardo, I2C). To get full readings I had to change in both `Wire/src/Wire.h` and `Wire/src/utility/twi.h` the `(TWI_)BUFFER_LENGTH 64` . 

[image]

paulvha, 2019-02-04:
great ! 
Indeed for Arduino & variants you have to change both. It is documented in the .odt file. On ESP8266 there is only one change needed and not needed on ESP32 (already 128). Still working to understand the full results and that is next point of focus. Hoping to learn from others here as well.

paulvha, 2019-02-08:
I have posted version 1.3.1 which fixes the issue that PM10 number is always showing zero's.

winkj, 2019-04-10:
@paulvha thanks for the work on the library. We've looked into providing an Arduino library, and have a version ready for AVR based Arduino's that supports reading both mass and counts. Currently, the ESP8266 is not supported yet, so I hope your library can help with that.

winkj, 2019-04-10:
@khongco44 there will be a fix for the ESP8266 in the next release of the ESP8266 Arduino libraries (version 2.5.0). For now, it is not officially supported yet. In the meantime, there exist third party solutions like @paulvha's that may be able to address your concerns.

I renamed this to be a bit more descriptive, and will update once the fix is in place.

systembolaget, 2019-09-16:
> That would be also be beneficial to Sensirion as AQ communities like [LuftDaten](https://luftdaten.info/) are constantly on the lookout for better/more accurate PM measurements.

That really is the issue. I want to use the sensor in a large citizen's science air monitoring project in Sweden, but to get interested citizens excited and to participate, a working I2C driver is necessary, because not every citizen can delve into the depths of the great work you guys are accomplishing here.

paulvha, 2019-09-16:
@systembolaget : if you are looking for a working I2C implementation for the SPS30, please check my library https://github.com/paulvha/sps30.

winkj, 2019-09-16:
@systembolaget, Sensirion provides a dedicated Arduino driver, which is available through the Arduino IDE Library Manager (library name is 'sensirion-sps') or for manual implementation from https://github.com/Sensirion/arduino-sps

winkj, 2019-09-16:
@abrauchli we should probably add a note to the README and close this

abrauchli, 2019-12-10:
ESP8266 should be fixed as of https://github.com/esp8266/Arduino/pull/3576 merged as part of 2.5.0.
Link to the arduino repo from the README is up for review.

closed 2020-01-22 by abrauchli

bonamibond, 2020-10-26:
Hi , 
I am trying to make SPS30 work with ESP8266. I got it working for Arduino Uno but it is not working on ESP8266 as it has only 3.3v output.
Can you please tell me how to fix the issue?

rnestler, 2020-10-26:
@bonamibond As mentioned here several time: Please use https://github.com/sensirion/arduino-sps if you work with an Arduino.