Sensirion/embedded-sps #16: reset Data-Ready Flag

michapr, 2019-01-19:
Hello,
I think after 
    0x0104 Stop Measurement Set Pointer
the Data Ready Flag should be resetted.
   (for 0x0202 Read Data-Ready Flag Set Pointer & Read Data)

In this time the flag show a "ready" and wrong results (not clear for my why worng) will be received.
In my case these values were much higher as the values before the "stop" command.

Maybe it is "cosmetic" - but it could help to get right values in case of program error ;)

abrauchli, 2019-01-21:
Hi @michapr 

The data ready flag has no meaning when the sensor is not in measuring state i.e. it's only relevant between `start measurement` and `stop measurement`. I'm also not sure what you're measuring after the `stop` command but since the sensor doesn't actually measure after `stop measurement`, I guess it'll report meaningless values.

I'm not quite sure if this answers your question, could you clarify otherwise?

michapr, 2019-01-22:
> it's only relevant between start measurement and stop measurement

yes, right. 
That's why I think that the data ready flag should be resetted to "not ready" after "stop measurement" command.

>  I'm also not sure what you're measuring after the stop command but since the sensor doesn't actually measure after stop measurement, I guess it'll report meaningless values.

-> that's why reset the **data ready** flag :)

abrauchli, 2019-01-22:
I see and I appreciate the feedback. From a user perspective I can somewhat relate but while I'm not involved in the firmware development for the SPS, code size is typically a big concern. As such I doubt we can consider a rather cosmetic fix like that. What you can of course do, is change the driver to report 0 in `get_data_ready` after you issued a `stop_measurement`, but it's likely not something we'd change in our drivers.

I'll close the issue as "won't fix".

closed 2019-01-22 by abrauchli