Sensirion/embedded-sps #18: I2C fails to initialise when using with nRF52 DK

nit1995, 2019-01-24:
The PM Sensor fails to initialise when using with nRF52 DK. I am using Segger Embedded Studio. The code keeps getting routed to NRF_BREAKPINT_COND in app_error_weak.c. This is the error I am getting. 

`Error 8: Initialization of I2C connection failed!
<info> app: PM Sensor Initialization started.
<error> app: ERROR 3735928559 [Unknown error code] at /home/dinesh/temp_data/nRF5_SDK_15.0.0_a53641a/modules/nrfx/drivers/src/nrfx_twim.c:250
PC at: 0x0002F8C3
<error> app: End of error report`

Any help would be appreciated.

rnestler, 2019-01-24:
@nit1995 Can you post the relevant parts of the code you are using? With just such a generic error message it's hard to tell what went wrong.
I guess you are using the [`Nordic_nRF5_series/sensirion_hw_i2c_implementation.c`](https://github.com/Sensirion/embedded-common/blob/ef40edfe74df5993c8868f1861d9e5b0ce1ab811/hw_i2c/sample-implementations/Nordic_nRF5_series/sensirion_hw_i2c_implementation.c)?

Did you verify that the [pin numbers](https://github.com/Sensirion/embedded-common/blob/ef40edfe74df5993c8868f1861d9e5b0ce1ab811/hw_i2c/sample-implementations/Nordic_nRF5_series/sensirion_hw_i2c_implementation.c#L42) match your platform?
>```cpp
>    /**
>     * Nordic specific configuration. Change the pin numbers if you use other pins
>     * than defined below.
>     */
>    #define SENSIRION_SDA_PIN 0
>    #define SENSIRION_SCL_PIN 2
> ```

nit1995, 2019-01-24:
Yes. I am using `Nordic_nRF5_series/sensirion_hw_i2c_implementation.c` and I changed the pin numbers to match the nRF52 DK board. 
I have included the sensirion_common.h, sensirion_arch_config.h, sensirion_common.h, sensirion_i2c.hand sps30.h. 
I have used the following examaple code: 
`static void test_sps30_pm_sensor(void)
{

  
    //struct sps30_measurement m; //make it global variable
    char serial[SPS_MAX_SERIAL_LEN];
    u8 auto_clean_days = 4;
    u32 auto_clean;
    u16 data_ready;
    s16 ret;
nrf_delay_ms(5000);
      // Start execution.
    NRF_LOG_INFO("PM Sensor Initialization started.");
     int x = sps30_probe();
      while (x != 0) {
        printf("PM Sensor Initialization failed\n");
        NRF_LOG_INFO("PM Sensor Initialization failed.");
    // sleep(1);
    }
    printf("PM Sensor Initialization successful\n");
    
    ret = sps30_set_fan_auto_cleaning_interval_days(auto_clean_days);
    if (ret)
        printf("error %d setting the auto-clean interval\n", ret);

    ret = sps30_start_measurement();
    if (ret < 0)
        printf("error starting measurement\n");
    printf("measurements started\n");
   // sleep(1);
   nrf_delay_ms(1);

    //do {
        do {
            ret = sps30_read_data_ready(&data_ready);
            if (ret < 0)
                printf("error %d reading data-ready flag\n", ret);
            else if (!data_ready)
                printf("data not ready, no new measurement available\n");
            else
                break;
            //usleep(100000); /* retry in 100ms */
             nrf_delay_ms(100);

        } while (1);

        ret = sps30_read_measurement(&m);
        if (ret < 0) {
            printf("error reading measurement\n");

        } else {
            printf("measured values:\n"
                   "\t%0.2f pm1.0\n"
                   "\t%0.2f pm2.5\n"
                   "\t%0.2f pm4.0\n"
                   "\t%0.2f pm10.0\n"
                   "\t%0.2f nc0.5\n"
                   "\t%0.2f nc1.0\n"
                   "\t%0.2f nc2.5\n"
                   "\t%0.2f nc4.5\n"
                   "\t%0.2f nc10.0\n"
                   "\t%0.2f typical particle size\n\n",
                   m.mc_1p0, m.mc_2p5, m.mc_4p0, m.mc_10p0,
                   m.nc_0p5, m.nc_1p0, m.nc_2p5, m.nc_4p0, m.nc_10p0,
                   m.typical_particle_size);

        }

        ret = sps30_stop_measurement();
       // sleep(1);
        nrf_delay_ms(1);
  //  } while (1);

    return 0;

}`

michapr, 2019-01-27:
@nit1995 :  maybe it is any communication / buffer problem?

You can try my simple sketch - maybe it will help. you can here better look for the problem, maybe.
https://smarthome-work.com/blog/?particulate-matter-sensor-sps30-and-arduino

It is working with small buffer too - but then only for the first 5 values.

rnestler, 2019-10-02:
@nit1995 Could you get it to work? If yes I'll close the issue. If no is there a way we can support you?

closed 2019-12-10 by abrauchli