Sensirion/embedded-sps #19: understanding PM0.5 / PM1.0 / typical particle size

michapr, 2019-01-27:
Hi,
if I get as sample 100 #/cm³ for PM0.5 and 105 #/cm³ for PM1.0 - why the "typical particle size" then can be 0.53μm ?
Is this not a middle value? How it will be calculated? (if most particles <= 0.5μm )

Thanks!

rnestler, 2019-01-28:
Hi @michapr 

What do the other values for PM2.5, PM4 and PM10 show? All 105 I assume since you didn't mention them?

>  Is this not a middle value? How it will be calculated? (if most particles <= 0.5μm )

I'm not sure about the exact calculation, but I will forward your question on how to interpret the "typical particle size" value.

michapr, 2019-01-28:
As sample - actual: 30.75 / 35.2 / 35.4 / 35.54 / 35.4 #/cm³ -> 0.54μm

So I would assume that "typical particle size" is <0.5μm .

Thanks!

paulvha, 2019-02-08:
According to the datasheet: PMx defines particles with a size smaller than “x” micrometers (e.g., PM2.5 = particles smaller than 2.5 μm).

That means (taking the samples mentioned above):
30.75  have a size up to 0.5 um                       >>  avg. size impact = 30.75 * 0.5
35.2 - 30.75 have a size between 0.5 and 1     >>  avg size impact = (35.2 - 30.75) * 1
35.4 - 35.2 have a size between 1 and 2.5 um >> etc
Add the avg size impact values  ( 20.325) and divide by total = PM10 = 35.4) gives a calculated avg size of 0.57.

PM0.5 | PM1 | PM2.5 | PM4 | PM10 | avg size
-- | -- | -- | -- | -- | --
30.75 | 35.2 | 35.4 | 35.4 | 35.4 | 0.54
0.5 | 1 | 2.5 | 4 | 10 |  
15.375 | 4.45 | 0.50 | 0 | 0 | 20.325
  |   |   |   |   | 0.574152542372881

It is not a 100% fit. Maybe they apply different multiplier for size impact, maybe have more information in the sensor than exposed, maybe include a number of the previous measurements in the calculations to prevent the number jump up and down too much between the snap-shots. Maybe Michapr can find out.

I had a sketch running for 175 samples, sample every 3 seconds
The average for the 175 typical size was : 0,575451860465117 um 
The average for 175 calculated avg was :  0,583083779069767 um,
Thus a delta of  0,007631918604651 over 175 samples. The error margin of 1.33%. I can live with that.

michapr, 2019-02-08:
Thanks, In the meantime I came to the same calculations ;)
Have calculated about 2000 samples within a week - my error margin is also about 1.5%

If @rnestler agree with this he could close it ;)

Thanks.

michapr, 2019-02-08:
BTW: your calculation is not full correct ;)

> 30.75 have a size up to 0.5 um >> avg. size impact = 30.75 * 0.5
> 35.2 - 30.75 have a size between 0.5 and 1 >> avg size impact = (35.2 - 30.75) * 1
> 35.4 - 35.2 have a size between 1 and 2.5 um >> etc

up to 0.5 means between 0 and 0.5 - so it will be 30.75 * 0.4 (as sample, because most will be near 0.5)
for size between 0.5 and 1.0 the middle value is 0.75  .. etc.

PM0.5 	PM1 	PM2.5 	PM4 	PM10 	avg size
30.75 	35.2 	35.4 	35.4 	35.4 	0.54
0.4 	 	0.75  	1.75 	 4 	 	10 	 
12.3 	3.34 	0.35 	0 	 	0 	 	15.99
  	  	  	  	  	 	 	 	 	 	0.4517

So it is in a lot of cases a bit away from displayed value ;)
But sometimes it is ok :)

rnestler, 2019-02-13:
I got the following answer from our engineers:

> The typical particle size (TPS) is not a function of the other SPS30 outputs, but an independent output. It gives an indication on the average particle diameter in the sample aerosol. Such output correlates with the weighted average of the number concentration bins measured with a TSI 3330 optical particle sizer.
> Following this definition, lighter aerosols will have smaller TPS values than heavier aerosols.
The reactiveness of this output increases with the particle statistics: a larger number of particles in the environment will generate more rapidly meaningful TPS values than a smaller number of particles (i.e., clean air).

I hope this cleans things up for you :slightly_smiling_face: Maybe we should add this to the README.

rnestler, 2019-02-13:
I'll close this for now, feel free to reopen if you have any further questions regarding the typical particle size.

closed 2019-02-13 by rnestler