Sensirion/embedded-sps #55: Driver returning very strange readings.

monkeytronics, 2021-04-13:
Hi, I'm using the provided driver with pretty much the example usage. It works. I can read the firmware version (2.2), the device serial number. And I can apparantly read back the values. The problem  is the values it reads are impossible (over 3000 um/m3 in a home). As others have said, the PM2.5 and PM10 values are identical. This in fact leads me to believe it is working. Or at least, sort of. I'm also using the SCD30, SHT30, SGP40 devices. And they all seem to be fine. They are using hte same common librarys to manipulate bytes to floats. I'm working on an ESP32, which is little endian, but the conversion seems fine for the other sensors. Any assistance welcome.

winkj, 2021-04-13:
What is your supply voltage? I've seen similar high values when using the SPS30 with at 3.3V instead of 5V, e.g. here: https://github.com/Sensirion/arduino-sps/issues/20

monkeytronics, 2021-04-13:
It turns out that the power supply (from PC USB) wasn't holding the 5V. It was fine until I added the extra load, and I didn't think to recheck it. Duh! Occam's razor.

Also, would be useful to add the following few lines of code to the code example. Since we aren't really giving it a long run while testing, this doesn't get done. Might help to see nice accurate results while we are in early dev stage. 

```
// Start Fan Clean : spins it up for about 10s at full speed to blow all the crap out.
    ESP_LOGI(TAG, "SPS30 Fan Manual Cleaning Started");
    ret = sps30_start_manual_fan_cleaning();
    if (ret < 0) {
        ESP_LOGE(TAG, "error in cleaning cycle!\n");
    }  
    
    vTaskDelay(10 * 1000 / portTICK_RATE_MS);    /* wait 10s */
```

closed 2021-04-13 by monkeytronics