Sensirion/embedded-sps #56: Getting the values for PM2.5 and PM10.0 is giving me values of 5000+ when using the example application.

PopherTheG, 2021-09-21:
`main.c: PM2.5: 5645.83 PM10.0: 5645.83`

Above are the values for them I get. I was wondering what is the unit for this and is there still a conversion factor to make the results into μg/m3?

winkj, 2021-09-21:
The output is in ug/m^3, so no further conversion is necessary.

Those values are obviously very high... Are you by any chance powering the SPS30 with 3.3V?

PopherTheG, 2021-09-21:
@winkj Yeah I am powering it with 3.3V since the MCU I am using is also powered with 3.3V. Is there a workaround for this?

winkj, 2021-09-21:
I'm not aware of any workaround; the supported supply voltage range for the SPS30 is 5V +/- 0.5V.

That said, the SPS30 will work fine with the 3.3V logic levels, so all you need to do is add a 5V line for the supply, SCL/SDA don't need any level shifting

winkj, 2021-09-21:
I'll go ahead and close this as it's not a software/driver issue. If you have further questions on the SPS30 hardware, feel free to reach out to our technical support team at https://www.sensirion.com/en/about-us/support-center/.

closed 2021-09-21 by winkj